Beepiz/BleGattCoroutines #4: Publish the library on jcenter

LouisCAD, 2018-01-02:
Should be done ASAP this month

bj0, 2018-01-02:
You may want to add to your download comment "but it's made of only 4 Kotlin files, so you can already try it out in your project pretty easily."  That it also uses another one of your libraries and how to get it, otherwise you have to dig through your project looking at build.gradle files.

LouisCAD, 2018-01-02:
I agree, I will edit the doc today. Thanks for your comment!
Sent with Android 8.1

LouisCAD, 2018-01-02:
@bj0 I just edited download instructions in the README to avoid any future confusion.

LouisCAD, 2018-03-11:
More than a month late, but will come soon. Likely the coming week

LouisCAD, 2018-03-19:
Version 0.1.0 is published on bintray! Sync to jcenter has been requested and is pending.

In the meantime, you can add the bintray repo to the `repositories` closure in your `build.gradle`:
`maven { url  "https://dl.bintray.com/beepiz/maven" }`,
and use the dependencies right now:
`implementation 'com.beepiz.blegattcoroutines:blegattcoroutines-core:0.1.0'`
`implementation 'com.beepiz.blegattcoroutines:blegattcoroutines-genericaccess:0.1.0'`

LouisCAD, 2018-03-20:
jcenter sync complete! 🎉 🎉
The Download instructions in the README have been updated.

PR #13 (2018-03-20): Update download instructions following jcenter sync
Closes #4
closed 2018-03-20 by LouisCAD