Beepiz/BleGattCoroutines #5: Need to write a descriptor to effectively enable characteristic notifications?

LouisCAD, 2018-01-04:
Calling `setCharacteristicNotificationsEnabled(...)` may not be enough, it may need an additional specific descriptor write.
We need to confirm this and see if the library should change accordingly.

Useful links:
https://developer.android.com/guide/topics/connectivity/bluetooth-le.html#notification
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
https://github.com/googlesamples/android-BluetoothLeGatt/blob/master/Application/src/main/java/com/example/android/bluetoothlegatt/SampleGattAttributes.java#L27

bj0, 2018-01-04:
This is indeed the case for me.  And from several SO questions (ie: https://stackoverflow.com/questions/17910322/android-ble-api-gatt-notification-not-received), it seems to be the case for Android in general.

According to that BT doc this descriptor is required, so assuming it is present is probably fine.  Using the following helper function I successfully get notifications:

```
suspend fun GattConnection.enableNotifications(characteristic: BGC, enabled: Boolean = true): BGD {
    setCharacteristicNotificationsEnabled(characteristic, enabled)

    val descriptor = characteristic.getDescriptor(clientCharacteristicConfiguration)
            ?: throw IllegalStateException("Required Client Characteristic Configuration not found")
    descriptor.value = ENABLE_NOTIFICATION_VALUE
    return writeDescriptor(descriptor)
}
```

with
```
val clientCharacteristicConfiguration = UUID.fromString("00002902-0000-1000-8000-00805f9b34fb")
```

LouisCAD, 2018-03-20:
There's interesting information about this [here on StackOverflow](https://stackoverflow.com/questions/22817005/why-does-setcharacteristicnotification-not-actually-enable-notifications).

It seems the `setCharacteristicNotification(…)` method only allows the Android device to receive notifications, but it doesn't mean they are enabled on the other side.

The `setCharacteristicNotificationsEnabled(…)`method from `GattConnection` will probably have its doc updated to state that fact, and may also be renamed (with deprecation cycle for the older name) to make it more obvious that it doesn't enable notifications on the other side.

Also, a suspending function to write that descriptor that enables notifications on the other side may be provided.

aravindsg, 2018-11-16:
@LouisCAD There are two notification types available in BLE. we need to write appropriate descriptor value to enable / disable BLE notification or indication.

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml&u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml

https://developer.android.com/reference/android/bluetooth/BluetoothGattDescriptor#fields

https://developer.android.com/guide/topics/connectivity/bluetooth-le#notification

Sample code
```
public static String CLIENT_CHARACTERISTIC_CONFIG = "00002902-0000-1000-8000-00805f9b34fb";

To Enable BLE Notification

mBluetoothGatt.setCharacteristicNotification(characteristic, enabled)
val uuid: UUID = UUID.fromString(CLIENT_CHARACTERISTIC_CONFIG)
val descriptor = characteristic.getDescriptor(uuid).apply {
    value = BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE
}
mBluetoothGatt.writeDescriptor(descriptor)

To Enable BLE Indication

mBluetoothGatt.setCharacteristicNotification(characteristic, enabled)
val uuid: UUID = UUID.fromString(CLIENT_CHARACTERISTIC_CONFIG)
val descriptor = characteristic.getDescriptor(uuid).apply {
    value = BluetoothGattDescriptor.ENABLE_INDICATION_VALUE
}
mBluetoothGatt.writeDescriptor(descriptor)

To Disable BLE Notification / Indication (OPTIONAL. it will disabled on GATT disconnection)

mBluetoothGatt.setCharacteristicNotification(characteristic, enabled)
val uuid: UUID = UUID.fromString(CLIENT_CHARACTERISTIC_CONFIG)
val descriptor = characteristic.getDescriptor(uuid).apply {
    value = BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE
}
mBluetoothGatt.writeDescriptor(descriptor)
```

closed 2019-05-05 by LouisCAD

LouisCAD, 2019-05-05:
Resolved in 0.4.0. Let me know if it works correctly for you or if you encounter any problem.

commit 2efe9bb5 (2019-05-05): Add notifications enabling on remote device

This commit includes the following changes:
1. The notifyChannel property now returns a new subscription each time,
so it is possible to have multiple consumers.
2. There is a new openNotificationSubscription function to receive
notifications only for a specific characteristic.

Resolves #5, resolves #12 and closes #41.