Beepiz/BleGattCoroutines #7: `connect` does oblivious to connection failure.

bj0, 2018-01-23:
If I try to connect to a device that is turned off, and wait with `connect().await()`,   I will eventually get a `StateChange` from `onConnectionStateChange` with a `status = 133 (GATT_ERROR)` and `newState = DISCONNECTED`, but the `await` will never complete.

I currently wrap it in a `withTimeout`, but it would be nice if the connection error threw on the `await`, so I don't have to wait for the full timeout.

bj0, 2018-01-23:
after some more tries, it looks like I don't get that error every time, which is pretty odd

LouisCAD, 2018-01-23:
What do you get instead when you don't have a 133 GATT_ERROR ?

bj0, 2018-01-23:
I was getting nothing.

LouisCAD, 2018-01-24:
This is expected since the device is not in range and powered on. The 133 GATT_ERROR is due to issues in the Bluetooth Stack however. As documented, it's your responsibility to handle connection status changes using the dedicated channel (so you can retry, or close the `GattConnection` object, cancelling the coroutines waiting on it). The sample shows how to properly use `withTimeout` to not let your app wait "forever". It allows you to specify exactly how much time you're willing to try to connect to a device that may no be in range at all.

LouisCAD, 2019-05-05:
Did using `withTimeout` solve your issue?