Beepiz/BleGattCoroutines #12: Notification Channel missing items

smasher816, 2018-03-20:
First of all, awesome library. Thank you so much for simplifying my bluetooth callback and handling code.

After enabling notifications and writing to the cccd as described here https://github.com/Beepiz/BleGattCoroutines/issues/5#issuecomment-355335034 I do sucessfully get notifications. However I discovered that notifications are sometimes not processed. Using the code below a log message seems to consistently only be produced for every other notification sent from the device.
```
notifyChannel.consumeEach {
    Log.i("BLEService", "notification: ${it.uuid} => ${it.value.toHexString()}")
}
```

Using the "nRF Connect" app every notification is received and displayed, so it does not appear to be a device/firmware issue.

LouisCAD, 2018-03-20:
Hi @smasher816, thanks for the kind words.
Do you listen to the `stateChangeChannel` to watch for potential errors or connection loss? Also, is nRF Connect active in the background when you encounter this issue?

LouisCAD, 2018-03-20:
The issue here is that by calling `receive()` on the `notifyChannel`, you suspend the coroutine until a notification is received. That means you don't reach the point where you subscribe to the other notification until a first one is received.

What you should do instead is adding a subscriber to the `notifyChannel` by calling `consumeEach { ... }` before applying any filter, and if you don't want to suspend the coroutine, do it in another coroutine (that you launch using `launch(coroutineContext)` or `async(coroutineContext)`.

LouisCAD, 2018-03-20:
@smasher816 Can you confirm that writing a descriptor is **required** in order to receive notifications, as stated in #5?

smasher816, 2018-03-20:
I do actually mean for the coroutine to suspend. My ble device has a kind of call and response architecture where the phone can pass it data, wait for human interaction, then compute a result and pass it back to the phone with a notification.

I have the `enableNotifications, write, notifyChannel.receive()` process above wrapped in a function called `writeAndReadResponse`. If I call `var foo = writeAndReadResponse(char1)`, then foo is set after the first notification sent from the device. If I then later want to compute another result and call `var bar = writeAndReadResponse(char2)` then the next notification sent from the device is ignored and it is not until I modify the code to send two notifications that the receive returns and bar is set. Sorry if that is a bit confusing.

I'll test #5 later today, but I know that before I migrated to your library I definitely had to write to the descriptor. Just calling BluetoothGatt.setCharacteristicNotificationsEnabled was not enough. See https://stackoverflow.com/questions/22817005/why-does-setcharacteristicnotification-not-actually-enable-notifications

LouisCAD, 2018-03-20:
Thanks for the link. It seems the `setCharacteristicNotification(…)` method only allows the Android device to receive notifications, but it doesn't mean they are enabled on the other side. That's probably the cause of your issue.

I commented in #5 on what we can expect in next version to address this.

LouisCAD, 2019-05-05:
This may have been caused by the fact that `notifyChannel` was not a `BroadcastChannel`. Version 0.4.0 just released should resolve your issue, and also brings a new `openNotificationSubscription` function that may be a better fit for what you want.

commit 2efe9bb5 (2019-05-05): Add notifications enabling on remote device

This commit includes the following changes:
1. The notifyChannel property now returns a new subscription each time,
so it is possible to have multiple consumers.
2. There is a new openNotificationSubscription function to receive
notifications only for a specific characteristic.

Resolves #5, resolves #12 and closes #41.
closed 2019-05-05 by LouisCAD