Beepiz/BleGattCoroutines #14: Feature request: Ble devices scan

jishindev, 2018-04-11:
Hi @LouisCAD, I've just started exploring kotlin co-routines and I'm really into learning more of it. I've written an app to interact with a ble device using the standard android SDK first and got frustrated with it so then used RxAndroidBle library and which does the job. But I assume that it all can be simplified using kotlin co-routines and your library. I checked this library and found it to be very well written. I have an app that requires to be able to scan for a particular type of ble devices and connect to one of them based on the rssi values or other criteria like mac address or device name.

Is it possible to have an api which emits all the devices that are found and then connect to one based on some criteria? May be have support for observing connection changes and also permission handling from within the library would make this a goto library for any one who's out to do fast prototyping. 

Thanks in Advance.

LouisCAD, 2018-04-11:
Hi @jishindev!
While how to design a library that exposes Android's `BluetoothGatt` API as suspending functions was pretty obvious, this was not the case for `BluetoothLeScanner`.
We use the scanner in our apps, but the code is really closer to our business logic, hence why it's not open sourced.

However, I know making bluetooth low energy scanning easier, and safer (because there are [undocumented pitfalls](https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983) in Nougat, and [Android 8.1 too](https://issuetracker.google.com/issues/71736547)) is something that you would want, and that we would like to have too. This would allow us to simplify our code that is currently mixed with business logic.

Asking for permissions is out of the scope of this library though (except for its sample).

Could you describe the complete lifecycle of what you do in BLE? I would like to know when you need to start scan, on which event, then for how long, under which conditions you cancel the scan, then if you connect under some condition, for how long (or if determined by some event, or some operations completion), if there are also possible cancellations, when you disconnect, if you need to reconnect shortly after, and if your logic is tied to an Activity, a foreground service, a scheduled or recurring job.

This is in order to have a use case example different than ours, to help in the design process of the scanning feature.

Note that this is a feature that was already planned, but with no immediate investment. We had in the idea to make a separate library that would be called something like BleScanCoroutines, so if you only need scan, no need to bundle the gatt code in your apk.

jishindev, 2018-04-11:
First of all, thanks for the quick and detailed reply.

> Asking for permissions is out of the scope of this library though (except for its sample).

In case if this grows into an end-to-end solution for interacting with BLE devices, may be we could just have callbacks that tell you if a certain permission is not granted or Bluetooth/Location settings are not turned on. Just in case. Not a must have. 

> Could you describe the complete lifecycle of what you do in BLE?

Sure. So we are working on a school bus attendance monitoring system with RFID. We have a Arduino Module which is used to scan the ID cards of students when the bus reaches their stops to alert parents that their kid is on-board or off-board.  This data from the Arduino has to be sent through BLE to  an Android app with the driver. This is done through one service with two characteristic where one is the student ID info from the RFID reader to the android app and the other is for validation data from the app to the Arduino saying the attendance was marked or not. The first characteristic has to have notification enabled to be alerted whenever a RFID card is read. 

> I would like to know when you need to start scan, on which event, then for how long, under which conditions you cancel the scan, then if you connect under some condition, for how long (or if [...]

The app flow is managed by a single activity having multiple fragments for each state of the trip. Since both the Android device with the driver and also the Arduino will be plugged in throughout the trip, we are not much bothered about the power efficiency of the connection process. Easier to just scan and connect to the device when the main activity is opened itself before proceeding further, and closing everything once the trip has ended. Or if we want to go the efficient way, this connection would made whenever the bus reaches a particular pickup or drop stop and then disconnected once the bus navigates to the next stop. The main reason to bring up this issue/feature request was that to eliminate the confusion during the connection phase if all the buses having these devices are in a close proximity lets say the school. 
Our logic is actually tied to a foreground service since we do let the app navigate the user to use Google Maps turn-by turn navigation which puts our app to the background.

Please let me know if you want anymore info. 
Thanks again.

LouisCAD, 2018-04-12:
@jishindev Is the Android device an Android Things device? Which Android version is it running?

LouisCAD, 2018-04-19:
@jishindev FYI, I'm starting to work on a library for scanning as we need it internally.

jishindev, 2018-04-19:
@LouisCAD Great. Thanks.

closed 2018-04-20 by jishindev

LouisCAD, 2018-04-20:
@jishindev Here it is: https://github.com/Beepiz/BleScanCoroutines
It's still in design phase, so there's no jcenter release, you'll have to download the project and import the core module to use it.