Beepiz/BleGattCoroutines #15: Update sample

jishindev, 2018-04-19:
Hi again @LouisCAD, Would be great if you could spare some time to update the sample android application with examples to read and write characteristics and descriptors and also notifications within the same sample repo.

LouisCAD, 2018-04-20:
Hi @jishindev,
We didn't do so yet because we would need to have a server BLE device that respects a contract that the client device (and app) would expect, and there's a lot of variety of BLE devices.

However, Android allows to open a GATT server, so here's what's possible:
- Optionally make a library module for GATT server to play well with coroutines.
- Make a GATT server in the `sample-common` module that implements most (or all) BLE operations, including notifications.
- Allow the mobile and wear sample apps to be used as a GATT server
- Make the GATT client part of the sample apps use this GATT server to showcase using BLE features.

Ideas are welcome, contribution is too, but discussion is needed first in order to get it right 😃

jishindev, 2018-04-21:
Valid point. But for now, may be we could have the example implementation mentioned [here](https://developer.android.com/guide/topics/connectivity/bluetooth-le.html) just to start with.

LouisCAD, 2018-04-21:
Yeah, it'd be a good start. PR welcome. Implementation is expected in the sample-common module, in dedicated package(s)