Beepiz/BleGattCoroutines #16: Request MTU

Hulcsa, 2018-05-09:
Hi!

First of all, thank you for this lib, very helpful! I only have one question.

Can I increase somehow the MTU size?

Bests,
Balazs

LouisCAD, 2018-05-09:
Hi @Hulcsa !

Thanks for the kind words!

This has not been included because we didn't need it and I wasn't certain how to expose it properly.
However, this is definitely something we want to add.

[The doc here](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback#onMtuChanged(android.bluetooth.BluetoothGatt,%20int,%20int)) states that the MTU changed callback can happen after a connection and after a change request. Consequently, I'm not sure how we should expose it.

Could you tell me more about your use case and more importantly, the order of the operations (you connect, then what do you, when you want to change MTU, if you can change it multiple times, if you need to know current MTU first…)?

Hulcsa, 2018-05-10:
Well, I only need to set it once before any communication. If i could find out what is the current MTU that would be nice too.

LouisCAD, 2018-05-14:
Ok, so in next version, this will be added.
The MTU value will be cached so you can it at any time

LouisCAD, 2018-11-26:
I won't be able to put this in the next version as I wrote previously, because my tests didn't exhibit the documented behavior regarding the `onMtuChanged` advertised to be called after a connection change.
I may provide an experimental API after Kotlin 1.3.10 migration though.

PR #38 (2019-05-05): Add requestMtu method to GattConnection
Fixes #16. I've tested this and it seems to work -- ended up being really easy to add since the requestMtu channel was already there (and previously unused?).

Let me know if there's anything else I should test, but it seems like this should be an easy merge. And then I can go back to using the mainline version of this library instead of my fork :)

Thanks for your work on this lib!
closed 2019-05-05 by LouisCAD