Beepiz/BleGattCoroutines #17: Add isConnected property on GattConnection

LouisCAD, 2018-05-14:


LouisCAD, 2018-05-22:
You're right @bj0, it already exists! I forgot it I had already added it. You say it's not being updated correctly… Does the `stateChangeChannel` receives any disconnection value?

closed 2018-05-22 by LouisCAD

bj0, 2018-05-22:
it does receive a `newState`, but the `status` flag is `8` when it happens, so it bypasses the `when` check.

LouisCAD, 2018-05-22:
@bj0 Out of curiosity, what device, running which version of Android are you using that gives you an `8` status?
Also, this may be an error code returned by the BLE device, you can check if in the same conditions but on a different device, you get the same status code

bj0, 2018-05-22:
It's a Samsung S7.  I don't think it comes from the BLE device as I get it several seconds after i turn the BLE device off.  I am assuming it has something to do with the connection timing out.

LouisCAD, 2018-05-23:
Strange...