Beepiz/BleGattCoroutines #28: androiid 23

aravindsg, 2018-11-16:


aravindsg, 2018-11-16:
@LouisCAD 
bluetoothDevice.connectGatt(appCtx, false, callback) doesn't work with latest Android api level.

We managed to get it work with this fix

```
    private val gatt: BG by uiLazy {
        if (Build.VERSION.SDK_INT >= M) {
            bluetoothDevice.connectGatt(appCtx, false, callback, BluetoothDevice.TRANSPORT_LE)
        } else {
            bluetoothDevice.connectGatt(appCtx, false, callback)
        }
    }
```

Reference:
https://stackoverflow.com/questions/50301602/unable-to-connect-android-to-a-specific-ble-device/52772343#52772343
https://stackoverflow.com/questions/18052593/how-to-connect-android-device-to-an-ios-device-over-ble-bluetooth-low-energy/51107363#51107363

LouisCAD, 2018-11-16:
Hi, thanks for the report, and the links.
This depends on the device you want to connect to.
At the moment, we've only tested it with one type of BLE only devices, so it works for us on all API levels.
However, I'm currently working on supporting initial connection parameters so you can have transport specified, and that should fix this issue.

commit 48e83051 (2018-11-16): Add ConnectionSettings parameter to GattConnection

Closes #21, closes #28
closed 2018-11-16 by LouisCAD