Beepiz/BleGattCoroutines #29: Proposal: Rename GattConnection interface to something else

LouisCAD, 2018-11-16:
`GattConnection` is not the ideal name for 3 reasons:
- `Bluetooth` or `Ble` is missing from its name, while it's exclusively for Bluetooth General ATTribute.
- There's "Connection" in the name, but when an instance is originally created, the connection is not
established yet.
- There's "Connection" in the name, but there's a `disconnect()` function… should `Connection` be magically dropped from the name of the class when this function is called? Of course, this is impossible, but shows how the name is not an ideal fit.

Consequently, I want this class to be renamed when the first Kotlin 1.3 compatible version is released, with a deprecated typealias on initial pre-release (alpha, beta or rc) for source compatibility.

Here are a few name ideas:
1. `BleGatt`
2. `SuspendBluetoothGatt`
3. `SuspendBleGatt`
4. `BluetoothGatt` (with named import to reference Android's as `AndroidBluetoothGatt` or alike)
5. `BluetoothGattIsFun` (wait… maybe not `actual` `fun` yet 🤔)
6. `FunBluetoothGatt` (this could open the discussion of a library renaming)

**Please, add a comment to say what you think about each proposition (I may append more name ideas, but I will keep the numbers stable), and if you have other name ideas, please write them there too.**

bj0, 2018-12-06:
Seems odd to put `suspend` on the front of the name, I would think `async` or `asynchronous` would be more descriptive/accurate.  Maybe `Allochronic`...

`BleGatt` is probably best but least unique.  Maybe `AndroidBleGatt Library`.  That gets close to `Able` though, which is used by (https://github.com/JuulLabs-OSS/able).

Or maybe just some sort of pun involving Cats?

Or just a non-descriptive name? `Bastet`?

es0329, 2018-12-18:
Hi, I'm passing through :) Given these options I agree with @bj0's favor/caution with regard to `BleGatt`. I also see on Wikipedia that [Bluetooth was first called "short-link"](https://en.wikipedia.org/wiki/Bluetooth#Origin) radio technology.

How does `interface ShortLink` sound to you?

```
internal class ShortLinkImpl( 
  private val bluetoothDevice: BluetoothDevice,
  private val connectionSettings: ShortLink.ConnectionSettings
) : ShortLink, CoroutineScope
```

JoeBakalor, 2018-12-18:
Short-link was in regards to Bluetooth classic though not Bluetooth Low Energy

WebTiger89, 2019-07-15:
> BleGatt sounds good

odahcam, 2020-01-17:
You could create a `GattConnectionBuilder` that returns the `GattConnection`, it would give time for ya to think in a library rename.