Beepiz/BleGattCoroutines #40: Connection maintained after timing out on connect() and calling close()

wcauchois, 2019-02-19:
Hi,

In the documentation for connect() it recommends you put a 5 second timeout on connect(), calling close() afterwards. I have followed this recommendation, but it seems I get in to circumstances where the connect times out but the connection to the device is maintained (just untracked by my code, which is the problem.)

My code is essentially:

```kotlin
val connection = GattConnection(device) // Note: autoConnect=false (the default)
try {
  withTimeout(5000L) {
    connection.connect()
  }
} catch (e: Exception) {
  connection.close()
  throw e
}
```

However, sometimes (actually quite frequently) after a timeout occurs I see that the device is still connected when I call `BluetoothManager.getConnectedDevices(BluetoothProfile.GATT)`.

There are lots of logs and weird messages from the Android BLE stack but I don't know what to make of them. Here are some logs of a repro. The affected device is `24:0A:C4:85:FD:4E`, so just search for that to find the relevant messages.

https://gist.githubusercontent.com/wcauchois/c5205e0c009e4f4a1fe31aa74cdcd22a/raw/ca7e766a8893cdafce745db776ff3dc2ed4dd635/ble_logs.txt

The "Connected devices:" log message prints the output from getConnectedDevices. At around 10:42:13, that "error handling scan result" message is where the timeout on connect exception is thrown. Above that it says "closing connection" which is the connection.close(), and you can see some system messages from BluetoothGatt seemingly saying something is closing. At 10:42:33 you can see the device still shows up in scan results. However at 10:42:34, we get some onServerConnectionState() message as well as "aclStateChangeCallback: device is NULL, address=24:0A:C4:85:FD:4E, newState=0". Subsequently at 10:42:38 our "connected devices" log message says that the device is connected.

So seemingly despite the close(), Android was still connecting somehow for around 21 seconds -- and ended up establishing the connection without our knowledge.

My device I'm testing on is a Pixel 3. Happy to provide any more info. Also, this may be just Android-level wonkiness and I should go to Stack Overflow, but would *really* appreciate any help on this! Thanks!

LouisCAD, 2019-02-28:
Hi Bill, thanks for reaching out regarding your issue, and sorry for replying this late.
I'd advise to raise the timeout to 7 seconds, or even a bit longer as in my experience, it can be quite slow at times.
That doesn't solve the underlying issue though, and in the current state, it's fair to say it's hard to debug it.

Something that could help debug is a debug mode that would log each call made on the `GattConnection` interface. You can already do it yourself by wrapping it to log and delegate to the provided implementation if you're in a hurry. That's something I plan to provide as a separate module in this project soon though. What do you think?

LouisCAD, 2019-05-05:
Probably related to https://github.com/NordicSemiconductor/Android-BLE-Library/issues/104

I believe this is caused by `connect` continuing despite the `close`, and should be solved by calling `disconnect`. I made `close` in `GattConnection` call `disconnect` by default, so this should solve your issue.

Next version will be out when this issue is automatically closed.

Please, let me know if it solves your issues as you expect.

commit 48029a78 (2019-05-05): Automatically trigger disconnection on close by default

Resolves #40
closed 2019-05-05 by LouisCAD