Beepiz/BleGattCoroutines #45: What about callbacks from gatt?

MikhailNatalenko, 2019-05-17:
I am sorry but i can't find anything about GATT's callbacks like `onConnection`, `onRecieve` and etc. How to us it?

MikhailNatalenko, 2019-05-17:
Ok, I don't need `onConnect` event because of coroutines, i just use it "synchronuously". But server can send me a messages sometimes and how should i get it?

LouisCAD, 2019-05-17:
The whole point of coroutines is to not pollute your code with routine callbacks. Or one of its points at least.

Also, I bet that by ""synchronously"", you meant "sequentially".

What are the messages sent from the server you are talking about? If it's notifications, this is already supported, you can find find all the documented functions you should need in the `GattConnection` interface.

MikhailNatalenko, 2019-05-17:
> Also, I bet that by ""synchronously"", you meant "sequentially".

@LouisCAD you right. English is not a native language for me...

I guess my question is pretty stupid. Should i get notifications like this:

```
//somewhere in my code
launch {
    while (true) {
        var notify = notifyChannel.recieve()
        //do something with notify.
    }    
}

```

MikhailNatalenko, 2019-05-17:
I think i begin to understand the main idea of corutines in BLE. I have a scenario:
```
connect
send data to characheristic AA
read data from AA
wait notify from BB
read BB
disconnect
```

i think i understand how to implement it with API

But what if i have a special characteristic CC that can send me a notify asyncronously at any time of my scenario? In my case this characteristic says about ERROR, and it sends me a ERROR-code, after that device waits for timeout and reboots. So it's very important to read CC in time to know the reason of reboot. How to read CC? I suppose i can't do it in any parallel coroutine

MikhailNatalenko, 2019-05-17:
Also how to add events from UI to this chain? The only way i see now is to have a synchro context with some values, which my Main Coroutine will read.  But sounds not good. Can you suggest another aproach?

LouisCAD, 2019-05-18:
If notification are enabled on both devices for the target characteristic, yes, `notifyChannel.receive()` in a loop can work, but please, look at [the openNotificationSubscription function](https://github.com/Beepiz/BleGattCoroutines/blob/2adfb4b0969478e60603f0abf1d506b03aac95eb/core/src/androidMain/kotlin/com/beepiz/bluetooth/gattcoroutines/GattConnection.kt#L165)

LouisCAD, 2019-05-18:
> The only way i see for now is to have a synchro context with some values, which my Main Coroutine will read.

What do you mean by "synchro context" and "some values"? And what are you referring to exactly when you write "Main Coroutine"?

MikhailNatalenko, 2019-05-18:
Ok , let me explain from the very begining. 

I have a BLE device, that sends me some events (with notify as you mentioned). I understand how does GATT works (in basics), i've already made this BLE device and i got an application that is made with callbacks. And it works pretty well but...

I want to refactor my BLE code with your API (because i want to try Kotlin and modern aproaches) and i don't understand how to design my app properly. 
I've got event-driven system. My BLE device can sand me various events time to time, so i have to catch them in separate infinite loop as we dicussed it before. But also i've got some events from UI, user can send cancel or something else to BLE device. So i should synchronise somehow UI thread and this loop. Also there is some buisness logic, like handshaking and validation between smartphone and BLE device (our special protocole). 

Is it possible to do protocole, catch events from BLE and handle user's command (like "Cancel") at the same time without synchronization of contexts and Spaghetti code? 

BTW proper and clear synchronisation is a main goal that i want to achieve with 

Thank you for help anyway

LouisCAD, 2019-05-18:
You can launch child concurrent coroutines with… `launch { }`. If you want to cancel some running coroutine based on user action or whatever, you can call `cancel()` on the `Job` returned by `launch`, as usual with coroutines. I'd recommend to launch a coroutine for listening to notifications, and you can launch other coroutines if it makes sense for your use case.

Also, if you want to abstract technical details away, you can write an `interface` with suspending functions and regular functions if needed to, that you implement using this library. That may help making your code cleaner and more independent from the library used, which is at least one step farther from spaghetti code.

Does it make sense to you? Let me know how that works for you and your codebase :)

MikhailNatalenko, 2019-05-19:
It really does thank you! Tomorrow i will try to do something with it.

MikhailNatalenko, 2019-05-20:
Also for those who has similar beginner's questions as me i recommed this video: 

https://www.youtube.com/watch?time_continue=2369&v=a3agLJQ6vt8 

Speaker tells how to use coroutines with channels, how to syncronise content and so on. It was really helpfull for me.

MikhailNatalenko, 2019-05-21:
So, i implemented everything i want with Channels

I've made bunch of channels, like 

`val eventAChannel : Channel<bool>` 
`val eventBChannel : Channel<bool>`

In main code, where i want to implement main functionality i do something like: 

```
...
eventBChannel.receive() // We will wait here our event
doMyActions()
eventAChannel.receive()///
fooBar()
...
```

Channels are syncrhonised out of the box so i don't worry about concurency and other async stuff

Then eanywere in code i can `send()` anything and i will get everything i want

I like this aproach very much.

MikhailNatalenko, 2019-05-21:
Thank you guys for this library!

closed 2019-05-21 by MikhailNatalenko