Beepiz/BleGattCoroutines #46: More verbous characteristics notify API

MikhailNatalenko, 2019-05-24:
Is it a good idea to make List of channels for each characteristic? (Is it even posible?) 

Now we got only one channel `notifyChannel`, when a notify comes we get it and then parse it (which craracteristic it is and so on), it is not very handy 

It would be nice if there were something like :

```
val deviceAnswer: RecieveChannel<byte[]> = setCharacteristicNotificationsEnabled(anserChar, true)
askSomething()
parseData(deviceAnswer.receive())
```

LouisCAD, 2019-05-25:
You already get a list of all the characteristics with `discoverServices()`, and you can then use `openNotificationSubscription`.

Does it satisfy your use case?

MikhailNatalenko, 2019-05-27:
No, i'm talking about API for recieving notifies (not enabling). For example i've got 2 different characteristics that can send me notifications. I will wait for notifications with `notifyChannel.receive()`  and then i should check which characteristic has sent the notification. It is ok for now, and i can use this API for my cases, but it would be much more handy if i could register separated notifyChannel**s** for different characteristics. It would allow us to use `select` for example.  Maybe it's a duplication of information but as for me i should make these channels anyway and think about async coroutine that will get the notification, parse it, send it next to my custom notification et cetera

LouisCAD, 2019-05-27:
The function I mentioned does both. Enables notifications for a specific
characteristic and funnels the received notifications into the returned
channel.

MikhailNatalenko, 2019-05-28:
i'm a dumbass, sorry i mixed up `openNotificationSubscription` and `setCharacteristicNotificationsEnabled` Imdeed this fun is what i need

closed 2019-05-28 by MikhailNatalenko