Beepiz/BleGattCoroutines #49: How to catch fatal disconnect?

MikhailNatalenko, 2019-06-10:
It's a little bit strange, but I do not catch anything from BLE if I disconnect it hardly. For example, if I switch the device off. Or if I turn off Bluetooth on my phone. I do catch everything if I disconnect from phone "properly".

So for example, if I connect the phone to the device and then reset the device - nothing will happen until I'll try to send something.

MikhailNatalenko, 2019-06-10:
Another simple case: connect to ble, and then just switch Bluetooth off.

MikhailNatalenko, 2019-06-10:
It looks like I don't set timeout. Is it possible?

LouisCAD, 2019-07-14:
> It looks like I don't set timeout. Is it possible?
I'm not sure what you mean here. Only you can know if you set timeout or not. Also, it's unclear to me what you're talking about when you ask if it is possible.

closed 2019-07-14 by MikhailNatalenko

LouisCAD, 2019-07-14:
Fixed formatting of my comment above.
Is there a reason why you closed it without telling why?

MikhailNatalenko, 2019-07-14:
Sorry, I was a little bit busy. To be honest, I don't even remember exactly the point of this issue, I'm not working with the library and the project anymore, so I'm not sure did I solve it by myself or not. Sorry again!