Beepiz/BleGattCoroutines #50: Incompatible version of the Android Gradle plugin

pablodeymo, 2019-06-19:
Hello Louis and team,

I am using Android Studio 3.5beta4, and I am having this error trying to compile the repo.

ERROR: The project is using an incompatible version of the Android Gradle plugin.
Please update your project to use version 3.5.0-beta04.
Search in build.gradle files

Can you help solving this?

(I agree with you about the hard and big effort using vanilla Android BluetoothGatt API, and I am glad to know about your project to simplify this).

Thanks.

pablodeymo, 2019-06-19:
I've changed version of Gradle in buildSrc to:
"com.android.tools.build:gradle:3.5.0-beta04" and now is building

LouisCAD, 2019-06-19:
Feel free to submit a PR directly on master (as the develop branch is behind for now) to update it.
I'm closing this as this is solved for you, and pretty common in the Android dev world.

closed 2019-06-19 by LouisCAD