Beepiz/BleGattCoroutines #54: GattConnection.discoverServices() not working

farruhha, 2019-10-03:
GattConnection.discoverServices() not working at all. When it is called, my app is just stuck, the code after discoverServices() is not running at all.

LouisCAD, 2019-10-04:
Are you certain it's not just taking a long time to execute?
How much time have you been waiting for it?
Does it happen all the times?
Does it happen only on a certain device?

LouisCAD, 2019-10-21:
Hi @farruhha, can you give the details requested 17 days ago?

farruhha, 2019-10-29:
Hello, I am sorry for the late response. I think we can close the ticket. It was my custom made hardware that has a problem with connection. Library is awesome to use.

LouisCAD, 2019-10-30:
Alright, and thanks for the kind words.

closed 2019-10-30 by LouisCAD

odahcam, 2020-01-16:
I'm facing the same issue with a chinese developed hardware. It works with common bluetooth approach, but not with the lib's `discoverServices` method. 

These are the last logs I received:

```
D/BluetoothAdapter: STATE_ON
D/BluetoothGatt: connect() - device: D9:91:53:2D:07:B1, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=fbb21088-17ec-4c0d-984d-cc206c536873
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=D9:91:53:2D:07:B1
```

After that, the app gets stuck forever. The same happen with other devices like Mi Band 4 and Amazfit Bip Watch.

I've verified that the debugger makes up to the `android.bluetooth.BluetoothGatt#discoverServices` method used in the `com.beepiz.blegattcoroutines/blegattcoroutines-core/0.4.1/e8ec123fc34a29aa69bc96db1f0da40354f1534a/blegattcoroutines-core-0.4.1-sources.jar!/androidMain/com/beepiz/bluetooth/gattcoroutines/GattConnectionImpl.kt:161` `discoverServices` method, but after that I couldn't find anything because the pointer got into some comments instead of code. Again, these are the logs I've received from the Android device when connecting and trying to discover services:

```
D/BluetoothAdapter: STATE_ON
D/BluetoothGatt: connect() - device: 00:15:90:91:CE:C2, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=cd9fbbb0-71fb-445f-81cf-620fb44d83f3
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=00:15:90:91:CE:C2
D/BluetoothGatt: onConnectionUpdated() - Device=00:15:90:91:CE:C2 interval=16 latency=0 timeout=800 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=00:15:90:91:CE:C2 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=00:15:90:91:CE:C2 interval=16 latency=0 timeout=800 status=0
```

I have no clue on whats happening.

odahcam, 2020-01-17:
> What is "common bluetooth approach" exactly?

The one Android has documented that consists in creating a `BluetoothLeService`, which I believe is the most common approach.

> FYI, debugger is going into comments is because the code on the device doesn't match the source code that you have, and line of code numbers are consequently not pointing to the same place.

That's probably because I'm using JetBrains uncompiler to reach that deep in the code, I didn't downloaded the sources to play with yet.

> From the logs you show, I see status=133, which indicates a GATT_ERROR caused by any of the two devices, or the app, or another app interfering.

That's probably because that's some random device I've tried to connect to to generate the logs, I've mistaken put these as the chinese hardware logs which, in fact, are in the second snippet.

> Regardless of that, I think it's be easier that you create a new issue if you believe there's something related to this library, because this issue was finally found to not really be related to this library.

I agree, I'll play with it tomorrow and see what happens.

Thanks for the quick response.