Beepiz/BleGattCoroutines #62: Update project

LouisCAD, 2020-03-23:
1. Integrate [refreshVersions](https://github.com/jmfayard/refreshVersions)
2. Update dependencies with it
3. Replace exposed `Channel`s with `Flow`s
4. Review PRs, and merge the ones that are improvements giving no drawback, or build upon the idea to make something better
5. Release a new version

Computr0n, 2020-03-28:
[image]

When cloning this repo with the latest stable Android Studio, I get that prompt and the automatic upgrade fails:
[image]

Any ideas?

francismariano, 2020-08-03:
Hello @LouisCAD 

Let us know about update project. Do you need help with something?

LouisCAD, 2020-09-14:
Hi @francismariano thanks for offering help and sorry for the long time with no response.

I was sidetracked by other open source projects ([refreshVersions](https://github.com/jmfayard/refreshVersions) and [Splitties](https://github.com/LouisCAD/Splitties)) and life.

I'll get back to this project and update it soon, before Kotlin 1.4 Online Event, and going forward, it'll be updated more often where needed as I'll add a few things to ease maintenance on my side (refreshVersions and CI).

Thanks for your patience.

LouisCAD, 2022-02-28:
Was much longer than I wanted, but now, the project has been updated. This issue is now obsolete 🙂

closed 2022-02-28 by LouisCAD