Beepiz/BleGattCoroutines #65: Migrate away from jcenter

joharei, 2021-03-01:
You're probably aware of this, but [jcenter is being sunsetted](https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/), so this package should probably migrate to somewhere else soon.

hannesvoss, 2021-06-27:
ping @LouisCAD

gapfDev, 2021-12-14:
Are you moving to a different library guys? ,   @joharei , @hannesvoss

joharei, 2021-12-15:
I moved to [Kable](https://github.com/JuulLabs/kable), which seems well maintained.

gapfDev, 2021-12-15:
Thanks @joharei

hannesvoss, 2021-12-15:
> I moved to [Kable](https://github.com/JuulLabs/kable), which seems well maintained.

Me too 👍🏼

LouisCAD, 2021-12-16:
Hello, I'm sorry for having taken so much time to respond.

I made myself so busy with other projects that it's hard to find enough time to maintain this one, and I'm not super motivated when I know that [kable](https://github.com/JuulLabs/kable), which is working on multiple platforms, provides the same functionality.

Since jcenter keeps being accessible for existing versions, it's not really a priority to publish the same artifacts to Maven Central. However, if I end up finding the time to work on this project again, I will publish newer versions on Maven Central, for sure.

LouisCAD, 2022-02-28:
Hello, last week, I finally published [a new version of the library](https://github.com/Beepiz/BleGattCoroutines/releases/tag/v0.5.0), (1004 days after the previous).

It's available on MavenCentral, and there's a smooth pipeline for future releases now.

LouisCAD, 2022-02-28:
Thank you for your patience 🙏

closed 2022-02-28 by LouisCAD