Beepiz/BleGattCoroutines #66: Seems like all work has stopped on this, if that is the case, please just mark it that way

KennyGoers, 2021-03-24:
Seems like all work has stopped on this, if that is the case, please just mark it that way.

Thanks

LouisCAD, 2021-03-24:
Hello, I'm still contributing to open source, it's just a struggle to focus on many things at once.
As time passes, the pile of things to do increases. I had some code changes to do, but now I have to deal with jcenter closing and setup MavenCentral releasing too, which makes me procrastinate as there are other things I want to do first.

I don't plan to abandon this project, unless something with an objectively better API comes up and there's no reason left to keep changing this one.

twyatt, 2021-03-24:
I by no means want to take any attention away from this library, as it is a great foundation for tackling the incredibly complex world of BLE. I also understand @LouisCAD being busy, he makes numerous contributions to open source and the Kotlin community in general. Although, as a potential alternative, I have been actively working on [Kable](https://github.com/JuulLabs/kable) which solves similar problems as this library.

LouisCAD, 2022-02-28:
Work on this library resumed.
Last week, I published version 0.5.0 on MavenCentral with several modernizing updates, it was 1004 days after the previous release (😅).

Now that the project is easier to update and publish, I'll do my best to respond much quicker (within a week or less).

closed 2022-02-28 by LouisCAD