Beepiz/BleGattCoroutines #69: How to avoid BLUETOOTH_CONNECT permission check when API version under 31?

tr15tan, 2022-07-11:
I got lint check error due to the annotation `@RequiresPermission(Manifest.permission.BLUETOOTH_CONNECT)` which you added for Android 12.

I have to write some boilerplate code generated from the IDE smart suggestion to force use the essential API like this :
```
            if (ActivityCompat.checkSelfPermission(
                    requireContext(),
                    Manifest.permission.BLUETOOTH_CONNECT
                ) != PackageManager.PERMISSION_GRANTED
            ) {
                Log.i("tristan", "missing Manifest.permission.BLUETOOTH_CONNECT permission")
                // TODO: Consider calling
                //    ActivityCompat#requestPermissions
                // here to request the missing permissions, and then overriding
                //   public void onRequestPermissionsResult(int requestCode, String[] permissions,
                //                                          int[] grantResults)
                // to handle the case where the user grants the permission. See the documentation
                // for ActivityCompat#requestPermissions for more details.
                // return
            }
            viewModel.myBLEMethod()
```

Can you give any suggestion about how to avoid these unnecessary code, please?

tr15tan, 2022-09-24:
Well, i found add this annotation `@SuppressLint("MissingPermission")` above the method calling `viewModel.myBLEMethod()` is just enough~

closed 2022-09-24 by tr15tan

LakshminPrasanna, 2022-10-25:
Hi @tr15tan , this @SuppressLint("MissingPermission") worked.But when i try with (device?.connectGatt(this, false, mGattCallback)) app started crashing and asking BLUETOOTH_CONNECT permission.If i gave the permission then app is not connecting to the BLE device. Can you please help me here

tr15tan, 2022-11-07:
> 

hi @LakshminPrasanna, i am not sure what the real problem you are facing with, but i think you can start from these  directions:

1. confirm your device is fully support BLE function
2. check the [permissions](https://developer.android.google.cn/guide/topics/connectivity/bluetooth/permissions) which BLE asked for.
3. check all verbose and warning logcat printed by BleGattCoroutines library and system
4. check the context passing to `connectGatt` method